kriswiner/MPU9250 #463: Air mouse help needed

Tonymazzz, 2021-07-04:
Hello Kris!
First of all great work with this libraries that you provided us !!
I need your help please sir, I am trying to emulate an Air mouse with mpu9250 and esp32 Via bt, and i tried lots of librarys but I get lost in how I can combine all the sensors so that the line:

Mouse.move(vx, vy);

Be perfectly balanced and ready for use in pc video games,
I found yours now and looks really nice for that un triying,

any advicement for what if yours librarys mpu9250 do i should start my project and help me with my questions please?

Im really noob un this world but eager to learn :)

Thanks in advance Kriss!!

kriswiner, 2021-07-04:
Getting accurate velocity estimation from an IMU is really quite difficult.
I suggest you consider something like this
<https://www.tindie.com/products/onehorse/pmw3901-optical-flow-sensor/> or
this <https://github.com/kriswiner/PAW3902> instead...much easier.

Tonymazzz, 2021-07-04:
Thank you very much for your quick response Kriss!!
 I'm going to study it: D