kriswiner/MPU9250 #464: Long term reliability of MPU9255

SD-Jafari, 2021-07-19:
Hi and thanks for your code job.
I have a question about MPU9255 and will thank you for your time for your honest answering.
I decided to use MPU9255 for an application that needs to work long time like years.
In some of your comments, you said that the sensor may be broken on accl part. is that really may happen to a well soldered chips into modules like GY-9255? if yes, how much trust must be taken to use this sensor to being used for commercial products? for being clear, i don't want to customers return back product for broken sensor!
Thanks for your time.

kriswiner, 2021-07-19:
All MEMs sensors can be broken from high impacts. There is nothing
particular about the MPU925X. They are all just vibrating silicon plates
and so are fragile. But in normal use this should not be a concern.

On the other hand, the MPU925X has been out of production for more than a
year now so it is not a good choice for a commercial product. Furthermore,
I would never rely on a breakout board (i.e., GY-9255) for a commercial
product. Much better to design a custom solution whose components can still
be sourced and whose design and fabrication you understand and control.

sd-mi, 2021-07-19:
Thanks for quick replay.
the main reason for using breakout boards instead of chip itself is hand soldering difficulties of a 3x3mm QFN chip!
can you advise some good replacement for this chip? i only need accl and gyro. it will be great if be a hand solderable package like SO or etc. 
Thanks.

kriswiner, 2021-07-19:
LSM6DSM is the best 6 DoF accel/gyro available now. This is LGA. You could
test using this
<https://www.tindie.com/products/onehorse/all-st-motion-sensor-breakout-board/>
or this
<https://www.tindie.com/products/onehorse/chainable-10-dof-motion-sensor/>.

I am sure there are cheaper breakouts available elsewhere too.

sd-mi, 2021-07-19:
Thank you for your time.

closed 2021-07-19 by sd-mi