kriswiner/MPU9250 #468: Euler angle calculation is different from Wikipedia?

de-viant, 2021-10-03:
Wikipedia shows 
[image]

And in the code it is
`    yaw   = atan2(2.0f * (q[1] * q[2] + q[0] * q[3]), q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3]);   `

`    pitch = -asin(2.0f * (q[1] * q[3] - q[0] * q[2]));`
  
`     roll  = atan2(2.0f * (q[0] * q[1] + q[2] * q[3]), q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3]);`

it seems Euler angle calculation is different from the wikipedia displays. (Formulae for Yaw and Roll seem interchanged)

https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles

kriswiner, 2021-10-03:
Maybe they define roll and yaw opposite to the way we do? WHich one gives
the correct results in your device?

de-viant, 2021-10-03:
your convention gives better result.
In my device I have an MPU6050 and QMC5883L (Clone of HMC5883), which have their axes aligned (atleast the x, y axes not so sure about z axis) so I update as
  
MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f,  mx,  my,  mz).
instead of 
MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f,  my,  -mx,  -mz
as in the MPU6050HMC5883AHR example sketch.

It seems to work fine, the only thing that bothers me is 
[image]

The small correlation I observe of yaw with change with roll or pitch. But changing yaw doesn't change roll, pitch.