kriswiner/MPU9250 #471: recogonize AK8963_ADDRESS

vahidyazdnian13, 2021-11-09:
hi ! i used your study about MPU9250 and that was very useful. thank you in advance.

when i run the code :    byte d = readByte(AK8963_ADDRESS, AK8963_WHO_AM_I);  // Read WHO_AM_I register for AK8963
    Serial.print("AK8963 "); Serial.print("I AM "); Serial.print(d, HEX); Serial.print(" I should be "); Serial.println(0x48, HEX);

i don't get 0x48 and instead i get 0. can you help me out how should i solve this problem ?
i am using a MPU9250 BMP280 SPI IIC/I2C 10DOF Acceleration Gyroscope Compass 9-Axis Sensor Board Module GY-91 along with Arduino uno.

kriswiner, 2021-11-09:
Not sure, are you using I2C with the Arduino UNO as master? Do you see the
MPU6500 on the I2C bus at 0x71? Did you enable pass through I2C?

vahidyazdnian13, 2021-11-09:
thank you for replying ! 
yes i use arduino UNO as master and yes i can see the MPU9250 on 0x71 and the WHO AM I for that is working properly but not for gyroscope.
i didn't get what u mean by enabling pass through I2C can you explain more please ?

kriswiner, 2021-11-09:
From the initMPU9250 function
<https://github.com/kriswiner/MPU9250/blob/master/MPU9250_BME280_SPIFlash_Ladybug/MPU9250.cpp>
there is this bit:

// Configure Interrupts and Bypass Enable
// Set interrupt pin active high, push-pull, hold interrupt pin level HIGH
until interrupt cleared,
// clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
// can join the I2C bus and all can be controlled by the Arduino as master
// writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12); // INT is 50 microsecond
pulse and any read to clear
writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01); // Enable data ready (bit 0)
interrupt So check the data sheet/register map for the INT_PIN_CFG
register. There is a bit there that allows I2C pass through so the host can
communicate directly with the AK8193C.

vahidyazdnian13, 2021-11-10:
thank you so much ! i didn't consider this point.
i have one  more question. i am using the code in" MPU9250BasicAHRS" ,so i should add this point to it right? i am wondering if i should add a interrupt pin from MPU9250 to arduino or not ?

vahidyazdnian13, 2021-11-10:
> From the initMPU9250 function <https://github.com/kriswiner/MPU9250/blob/master/MPU9250_BME280_SPIFlash_Ladybug/MPU9250.cpp> there is this bit: // Configure Interrupts and Bypass Enable // Set [...]

now i realized that the point you mentioned was in the last part of "initMPU920" function so i think it does enable it but my wiring is like what is mentioned in the top of the arduino code. is that ok ? what more should i do to solve this problem ?