kriswiner/MPU9250 #475: 5 different MPU9250 and not a single one 100% working (broken axis)

simkard69, 2022-03-08:
Hello everyone,

I'm tryting to get this lib working, and I think it does pretty well except the fact that when I'm trying to read from each sensor I have (4x GY-91 and 1x GY-9250), not a single one seems to give 3 working axis for Accel and/or Gyro.
The 3 magnetometer axis are systematically ok (maybe much more resistant than the ones from accel/gyro).

I always get 1 axis or 2 not working per sensor on accel and/or gyro ... very frustrating.

Can someone help me out with those ? maybe it's just a calibration problem ... or maybe these sensors are just 5h177y ones systematically ?

Thanks

kriswiner, 2022-03-08:
Not sure what you mean here by one axis not working.

So you are getting data from one or two accel axes, lets say, but what no
data from the z-axis means all zeroes or can't read the register or what?

puredirect, 2022-04-08:
hi! i have the same problem, bought 4 mpu9250 and none of them works... some axis gives frozen data
have you solved the issue?
please write profaneus@gmail.com

kriswiner, 2022-04-08:
These either work or they don't. If you are getting reasonable data from
one of the six sensor axes and "frozen" data from the other axes then the
device is fubar.

Unless you have "fixed" the sketches somehow...

Take a look at the raw register bytes to see if these are changing or not.
If the register bytes do not change even as you move the sensor around the
devices are broken and should be thrown away.

I am not surprised that this is happening more now when the MPU9250 has
been out of production for more than two years and people are selling all
of their stale, crappy inventory.

puredirect, 2022-04-10:
Thank you! I bought the 4th one from TME and it is not even initializing…. Why i cant find any  information abot this scam…. Pls tell me what to buy. I would like to by a good gyro. With a good library…

Az iPhone-omról küldve

kriswiner, 2022-04-10:
ST's 6 DoF IMUs like the LSM6DSM, LSM6DSR are very good (better than
MPU6500), and there are breakouts
<https://www.tindie.com/products/onehorse/all-st-motion-sensor-breakout-board/>
available.

Invensense ICM42688 is pretty good too if you can find a breakout. Add a
good magnetometer like the MMC5983 and you would have a much better
replacement for the MPU9250.

There are also IMUs that do the fusion for you. This
<https://www.tindie.com/products/onehorse/usfsmax-module-and-carrier/> is
superbly accurate.