kriswiner/MPU9250 #478: Reading ACCEL and MAG Registers are wrong?

gawsalyan, 2022-05-21:


kriswiner, 2022-05-21:
Is there a question here?

gawsalyan, 2022-05-21:
Hi @kriswiner,

I find that the read operations are only performed in ACCEL_XOUT_H and GYRO_XOUT_H.
Why other Y,Z registers and _L (lower bit) registers are not used?

It seems like only ACCEL_XOUT_H register is read 6 byte and used in all? instead of ACCEL_XOUT_H, ACCEL_XOUT_L, ACCEL_YOUT_H, ACCEL_YOUT_L, ACCEL_ZOUT_H, ACCEL_ZOUT_L

Same applies to Gyro

e.g.

void readAccelData(int16_t * destination)
{
uint8_t rawData[6]; // x/y/z accel register data stored here
readBytes(MPU9250_ADDRESS, ACCEL_XOUT_H, 6, &rawData[0]); // Read the six raw data registers into data array
destination[0] = (int16_t)(((int16_t)rawData[0] << 8) | rawData[1]) ; // Turn the MSB and LSB into a signed 16-bit value
destination[1] = (int16_t)(((int16_t)rawData[2] << 8) | rawData[3]) ;
destination[2] = (int16_t)(((int16_t)rawData[4] << 8) | rawData[5]) ;
}

kriswiner, 2022-05-21:
This function:

readBytes(MPU9250_ADDRESS, ACCEL_XOUT_H, 6, &rawData[0]); **

reads six bytes, starting from the ACCEL_XOUT_H register and incrementing the register address until all six bytes for the accel are read and stored in the rawData buffer.

It does not read the same register six times. Is this what you are asking?

gawsalyan, 2022-05-21:
Yes, Thanks a lot for the clarification :)

kriswiner, 2022-05-21:
Most sensors have the ability to automatically increment the register
address. For some, like many ST sensors, one has to set a bit in a control
register to allow this to happen. For Invensense (TDK) sensors, this is
done automatically.

gawsalyan, 2022-05-21:
Yes, Thanks again for the clarification, I didn't realize that we are reading a contiguous memory :)