kriswiner/MPU9250 #479: MPU9250 custom board didn‘t work

Lambertttt, 2022-06-21:
Hallo Kris,
I'm working on my research project and I made a custom board with the ESP32 chip, it also includes some sensors like: MQ4,MQ7, TTP223B, MPU9250 and so on. When I get the board, the other sensors all works fine except the MPU9250, I'm sure it's not because of the mistake of the program cause before I make the board, I bought all the sensors I need and test so many times in the breadboard, so it can only be a problem with the circuit or the chip itself. I have attached my circuit diagram below along with the PCB diagram of my 2 layers board. Thank you in advance, any help will be useful!

I have summed up a few possible problems by myself through the multimeter and searching on Internet, but it's hard to test in a finished board, so could you please give me some advise for these possible problems, I will be very appreciated:
1. I searched some circuits for MPU9250 and there are always a voltage regulator chip like LB2K, there is a LM1117 on my board for all the sensors so I didn't add another voltage regulator chip only for the MPU9250, will this have some influence? I tested the voltage from SDA and SCL pin to GND pin, they are all 3.27V.
2. The next possible problem is about current. I checked the data sheet and I found the regular working current for the Gyroscope, Accelerometer and the Magnetometer are 3.2mA, 450uA and 280uA. I tested the circuit on bread board and my own board, the current through the SCL's pull up resistor(R4) is 20mA(Is this the working current for the chip?) for the tested circuit on the bread board and 18mA for my custom board. I also found that the current through R8(connect to FSYNC) is 68mA and R5(SDA), R6(nCS) and R7(AD0)'s current is zero. Are these currents' status correct?
3. I saw some issues on this page before, seems like you use 4,7k pull up resistor, should I change 10k to 4.7k?
4. Pls help me to check are there some other mistakes for my circuit. Are there some mistakes for the resistors and the capacities?
5. I test it with I2C scanner, it says: no I2C device found.

I will be very grateful for your help!

[image]

[image]

[image]

kriswiner, 2022-06-21:
I don't see any reason why it shouldn't work. Do you have GPIO22 and 21
specified properly in the ESP32 sketch?

Lambertttt, 2022-06-22:
Thanks for your response, I've checked the wiring and pins, it's not a problem.
I'm even starting to wonder if this chip is bad, do you have some ways to see if this chip is bad?

kriswiner, 2022-06-22:
MPU9250 is out of production for 18 months now so I wouldn't be
surprised if the ones used in your project were stored improperly or
otherwise compromised. Then there were a lot of fakes made too since it was
a popular and rather expensive chip in its day. Best way to tell is to look
at the package markings and compare with the data sheet.

I am not sure how you  measured the current but when the MPU9250 is working
at, say 200 Hz sample rate it uses ~10 mA. So if you are seeing ~50 mA
through some of its pins there is something very wrong,..

Do you have more than one board? If fake or bad MPU9250 were used in your
build it won;t matter, but if a cold solder joint or solder bridge were
present (assembly error) it might work on another board...

Lambertttt, 2022-06-22:
Thank you very much, I will test the other board.