kriswiner/MPU9250 #480: Alternative to MPU9250 and BNO055

Stephan-Cl, 2022-06-23:
Hello Kris,
Since you're an expert at that, I was wondering if you had any recommendation for an easy-to-use yet reliable alternative to the above chips, as those are no longer available. I'm looking for an I2C-based, ready-to-go board to interface with an ESP32. The chip should also have some sort of library (Arduino or such) to quickly get me going.
Any help appreciated.

Regards,

S. Clark

kriswiner, 2022-06-23:
I assume you are looking for 1) quaternions and/or Euler angles and 2) high
accuracy and 3) low power? This
<https://www.tindie.com/products/onehorse/usfsmax-module-and-carrier/> is
the best available (< 0.5 deg rms heading accuracy), this
<https://www.tindie.com/products/onehorse/max32660-motion-co-processor/> is
really good (also <0.5 degree rms heading accuracy but heat sensitive), and
this
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/>
is pretty good (<2 degree rms heading accuracy). All use a co-processor to
take the load off of the host MCU and are easy to use. Either is way better
than the BNO055 or MPU9250 DMP.

I have heard that the ICM29048 9 DoF IMU is pretty good (about the same as
the pretty good option above, but never used it myself). It has the
advantage of no co-processor needed but also requires a 1V8 digital rail,
so less convenient to use. But if you want something to add to an
existing or custom pcb this might be worth a look.

If you want to do Madgwick or Mahony fusion on the ESP32 host you could use
the LSM6DSM + MC5983; this is the best combo available although the
ICM42688 is also pretty good as a 6 DoF accel/gyro IMU. There is no mag
available better than the MC5983, but it does use a bit more power than,
say, the LIS2MDL, my second choice.

StayYouth1993, 2022-08-09:
What about MPU6500 + MC5983, which is much more cost down for me.

kriswiner, 2022-08-09:
This would be a good choice, but I still think the LSM6DSM/R is a better 6
DoF sensor. Instead of the MPU6500, mabe look at the ICM426X8?

kriswiner, 2022-08-09:
ICM42605 or ICM42688

StayYouth1993, 2022-08-09:
> This would be a good choice, but I still think the LSM6DSM/R is a better 6 DoF sensor. Instead of the MPU6500, mabe look at the ICM426X8?
> […](#)

I have inquiried the price(1k pcs) just now, which is:
1. LSM6DSRTR 9RMB
2. LSM6DSMTR 13RMB
3. MPU6050 5.5RMB(which is not totally new, price for totally new is 35RMB)
4. MMC5983MA 4.3RMB
As for ICM42605 and ICM42668, that's not avaliable in china.

I'm more cost sensitive. However, consider the chip quality, so my final choice is LSM6DSRTR + MMC5983MA. Thanks so much for your help!!!

StayYouth1993, 2022-08-09:
I also tested MPU9255(I have bought so many fake chip which actually is MPU6500, I have to give up on plan for MPU925x) with magwick fusion algorithm, which performance is very good without fast moving or shaking. However, when it comes for fast moving on human body(especially hands moving), magwick is not good enought, I think some strategy has to be made for this situation.

kriswiner, 2022-08-09:
For fast response you need a fast MCU like Cortex M4F with floating point
unit. I like the STM32L4, like this
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>
one.

StayYouth1993, 2022-08-09:
Oh, that's a good choice! I'll try later, thanks so much for your help!!!!!!!