kriswiner/MPU9250 #481: How can I get Ak8963 Address from MPU9250

Ibarguen, 2022-07-28:
I try to read the value Who I Am from AK8963 in MPU9250, if I read the values of the accelerometer and the gyroscope in other skecth I have not problems, but When I try to configure the magnetometer first I need to know if there is connection in i2c bus but, If I run i2c Scan always find 0x68-- MPU9250 address but never AK8963 Addres 0X0C, I have already configured Bypass from Int_cpf.. register Address 0x37 with 0x02 here my code always return 255 from value Who I Am from AK8963.

Here my code
```

#include <Wire.h>
     const uint8_t AK8963 = 0X0C;//
    const uint8_t WIA = 0X00;
    const uint8_t MPU9250 = 0x68;

     const uint8_t ACCEL_XOUT_H = 0X3B;//direccion del registro donde se aloja el valor de lectura 
      acelerometro

    const uint8_t SMPLRT_DIV = 0X19;//registro divisor de frecuencia 
    //la frecuencia de muestreo del sensor es igual a la frecuencia del giroscopio divido 
    //el valor  dado en el registro SMPLRT_DIV + 1
    //Frecuencia de muestreo  giroscopio output rate/(1+SMPLRT_DIV) = 8KHZ/8 = 1KHZ

const uint8_t CONFIG = 0x1A;//Registro de configuracion de tasas de muestro y filtro acelerometro y giroscopio
const uint8_t USER_CTRL   =  0x6A;
const uint8_t SIGNAL_PATH_RESET = 0X68;
const uint8_t FIFO_EN = 0X23;
const uint8_t INT_ENABLE = 0X38;
const uint8_t PWR_MGMT_1   =  0x6B;
const uint8_t PWR_MGMT_2   =  0x6C;
const uint8_t ACCEL_CONFIG = 0X1C;//registro configuracion acelerometro
const uint8_t INT_PIN_CFG = 0X37;

void MPU9250_CONFIG()
{

delay(150);//espero 100 milisegundos
I2CWRITER(MPU9250 , SMPLRT_DIV, 0X07);//Envio el numero 7 al registro 
I2CWRITER(MPU9250 , CONFIG, 0X00);//Giroscopio con frecuencia de muestreo de 8KHZ
I2CWRITER(MPU9250 , FIFO_EN, 0X00);
I2CWRITER(MPU9250 , ACCEL_CONFIG , 0x00);
I2CWRITER(MPU9250 , INT_ENABLE, 0X01);
I2CWRITER(MPU9250 , SIGNAL_PATH_RESET, 0x00);
I2CWRITER(MPU9250 , USER_CTRL , 0X00);
I2CWRITER(MPU9250 , PWR_MGMT_1, 0x01);
I2CWRITER(MPU9250 , PWR_MGMT_2, 0x00);
I2CWRITER(MPU9250 , INT_PIN_CFG, 0x02);

}
void I2CWRITER(uint8_t deviceAddress, uint8_t regAdress, uint8_t dato)
{
   Wire.beginTransmission(deviceAddress);  
   Wire.write(regAdress);
   Wire.write(dato);
   Wire.endTransmission();

}

void setup()

{
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
  MPU9250_CONFIG();
  
  
}

void loop() {

  Wire.beginTransmission(AK8963);
  Wire.write(WIA);  
  Wire.requestFrom(AK8963, 1);    
  byte c = Wire.read(); // receive a byte as character
  Serial.println(c);  
  delay(500);// print the character
  }
```

Datasheet say the register Who I am of AK8963 Magnetometer Should return 0x48 but I always read 0xFF or 255 in decimal..

kriswiner, 2022-07-28:
No idea here and I cannot debug your code. Please have a look at the
example sketches which work well. You should only have to set the I2C
bypass bit as part of the MPU6500 initialization to be able to talk with
the AK8963C. If this doesn;t work, maybe there is something wrong with the
IC, like it has no magnetometer....

tharunrp06, 2025-09-19:
> I try to read the value Who I Am from AK8963 in MPU9250, if I read the values of the accelerometer and the gyroscope in other skecth I have not problems, but When I try to configure the magnetometer first I need to know if there is connection in i2c bus but, If I run i2c Scan always find 0x68-- MPU9250 address but never AK8963 Addres 0X0C, I have already configured Bypass from Int_cpf.. register Address 0x37 with 0x02 here my code always return 255 from value Who I Am from AK8963.
> 
> Here my code
> 
> ```
> 
> #include <Wire.h>
>      const uint8_t AK8963 = 0X0C;//
>     const uint8_t WIA = 0X00;
>     const uint8_t MPU9250 = 0x68;
> 
>      const uint8_t ACCEL_XOUT_H = 0X3B;//direccion del registro donde se aloja el valor de lectura 
>       acelerometro
> 
>     const uint8_t SMPLRT_DIV = 0X19;//registro divisor de frecuencia 
>     //la frecuencia de muestreo del sensor es igual a la frecuencia del giroscopio divido 
>     //el valor  dado en el registro SMPLRT_DIV + 1
>     //Frecuencia de muestreo  giroscopio output rate/(1+SMPLRT_DIV) = 8KHZ/8 = 1KHZ
> 
> 
> 
> const uint8_t CONFIG = 0x1A;//Registro de configuracion de tasas de muestro y filtro acelerometro y giroscopio
> const uint8_t USER_CTRL   =  0x6A;
> const uint8_t SIGNAL_PATH_RESET = 0X68;
> const uint8_t FIFO_EN = 0X23;
> const uint8_t INT_ENABLE = 0X38;
> const uint8_t PWR_MGMT_1   =  0x6B;
> const uint8_t PWR_MGMT_2   =  0x6C;
> const uint8_t ACCEL_CONFIG = 0X1C;//registro configuracion acelerometro
> const uint8_t INT_PIN_CFG = 0X37;
> 
> 
> void MPU9250_CONFIG()
> {
> 
> delay(150);//espero 100 milisegundos
> I2CWRITER(MPU9250 , SMPLRT_DIV, 0X07);//Envio el numero 7 al registro 
> I2CWRITER(MPU9250 , CONFIG, 0X00);//Giroscopio con frecuencia de muestreo de 8KHZ
> I2CWRITER(MPU9250 , FIFO_EN, 0X00);
> I2CWRITER(MPU9250 , ACCEL_CONFIG , 0x00);
> I2CWRITER(MPU9250 , INT_ENABLE, 0X01);
> I2CWRITER(MPU9250 , SIGNAL_PATH_RESET, 0x00);
> I2CWRITER(MPU9250 , USER_CTRL , 0X00);
> I2CWRITER(MPU9250 , PWR_MGMT_1, 0x01);
> I2CWRITER(MPU9250 , PWR_MGMT_2, 0x00);
> I2CWRITER(MPU9250 , INT_PIN_CFG, 0x02);
> 
> }
> void I2CWRITER(uint8_t deviceAddress, uint8_t regAdress, uint8_t dato)
> {
>    Wire.beginTransmission(deviceAddress);  
>    Wire.write(regAdress);
>    Wire.write(dato);
>    Wire.endTransmission();
> 
> 
> }
> 
> 
> 
> void setup()
> 
> {
>   Wire.begin();        // join i2c bus (address optional for master)
>   Serial.begin(9600);  // start serial for output
>   MPU9250_CONFIG();
>   
>   
> }
> 
> void loop() {
> 
>   Wire.beginTransmission(AK8963);
>   Wire.write(WIA);  
>   Wire.requestFrom(AK8963, 1);    
>   byte c = Wire.read(); // receive a byte as character
>   Serial.println(c);  
>   delay(500);// print the character
>   }
> ```
> 
> Datasheet say the register Who I am of AK8963 Magnetometer Should return 0x48 but I always read 0xFF or 255 in decimal..
> 
> [#187](https://github.com/kriswiner/MPU9250/issues/187) [#123](https://github.com/kriswiner/MPU9250/issues/123) [#85](https://github.com/kriswiner/MPU9250/issues/85)
> 
> [@kriswiner](https://github.com/kriswiner) Please friend Help me, I try this from 1 week ago I',m very sad.. I can read Accelerometer and Gyroscope values in other sketch even Send the datas via rs232 or Wifi using ESP8266 Board but I can't comunnicate with AK8963.
> 
> So in the page 47 of register datasheet say this :
> 
> Addresses from 00H to 0CH and from 10H to 12H are compliant with automatic increment function of serial interface respectively. Values of addresses from 10H to 12H can be read only in Fuse access mode. In other modes, read data is not correct.
> 
> Maybe is that the problem ?

Did you resolve this?  I am too experiencing similar kind of issue.

tharunrp06, 2025-09-19:
> Not sure what to say here. I assume you are taking the usual steps to read
> the mag like verifying I2C works on the accel/gyro, placing the MPU6500
> into bypass mode, etc. There is no magic spell to make this work, just
> follow the data sheets and sample Arduino code.
>
> There have always been a lot of "fake" MPU9250s out there, more so since
> the MPU9250 went out of production 7 or 8 years ago. It is likely you have
> one of the Invensense variants that do not have a magnetometer, just the
> MPU6500 combo accel/gyro. Lots of unscrupulous (or ignorant) Chinese
> retailers sell these as MPU9250 since there is still high demand for this
> sensor.
>
> My recommendation is to buy an MPU9250 breakout board from a reputable
> seller like Sparkfun, Adafruit, etc (if you can still find one) to test
> and
> develop your code. I have had to throw away a fair number of "MPU9250
> sensors" I bought on Aliexpress that turned out to be something else....

Thanks for the response. My MPU9250 is from Invensense only. And I am sure
now that it does not have any magnetometer. Looking to buy a seperate
magnetometer or use the GPS itself for the Yaw control in my Low Level RTOS
controlled Drone. Thank you once again for your help!

kriswiner, 2025-09-19:
I recommend the MC5983MA, this is the mag we use.

kriswiner, 2025-09-19:
This <https://www.tindie.com/products/onehorse/usfsmax-module-and-carrier/>
is our latest offering as a reference. You won't find anything more
accurate.