kriswiner/MPU9250 #483: The Magnetometer adress doesn't show up at MPU9250 with RPI3

dritonkrasnici, 2022-10-17:
Hi,
Im only able to see the 0x68 Adress of the MPU6050 but not the 0x0c Adress of the Magnetometer. 
Im using two MPU9250 Sensor connected via I2C with a Raspi 3.

> Output after ` i2cdetect -y 1`:

  ------ 0 1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
   00:          -- -- -- -- -- -- -- -- -- -- -- -- --
  10:  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  20:  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  30:  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  40:  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  50:  -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  60:  -- -- -- -- -- -- -- -- 68 69 -- -- -- -- -- --
  70:  -- -- -- -- -- -- -- --

Do i have to set some Registers ?

kriswiner, 2022-10-17:
Yes, you have to set the master i2c bypass mode to on.

dritonkrasnici, 2022-10-29:
It still doesn't work.
I set the master i2c bypass like this: 

`bus1.write_byte_data(0x68,0x37,0b00000010)`
[image]

but it has no effect. 
I still cannot see the 0x0C adress on the i2c bus.

kriswiner, 2022-10-29:
Yup, maybe you have one of the "fake" MPU9250 that doesn't have the mag...

Just in case, here is my MPU9250 init API. After this, the AK8963C will
show on the I2C bus.

void initMPU9250()
{
 // wake up device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00); // Clear sleep mode bit
(6), enable all sensors
  delay(100); // Wait for all registers to reset

 // get stable time source
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01);  // Auto select clock
source to be PLL gyroscope reference if ready else
  delay(200);

 // Configure Gyro and Thermometer
 // Disable FSYNC and set thermometer and gyro bandwidth to 41 and 42 Hz,
respectively;
 // minimum delay time for this setting is 5.9 ms, which means sensor
fusion update rates cannot
 // be higher than 1 / 0.0059 = 170 Hz
 // DLPF_CFG = bits 2:0 = 011; this limits the sample rate to 1000 Hz for
both
 // With the MPU9250, it is possible to get gyro sample rates of 32 kHz
(!), 8 kHz, or 1 kHz
  writeByte(MPU9250_ADDRESS, CONFIG, 0x03);

 // Set sample rate = gyroscope output rate/(1 + SMPLRT_DIV)
  writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x04);  // Use a 200 Hz rate; a
rate consistent with the filter update rate
                                    // determined inset in CONFIG above

 // Set gyroscope full scale range
 // Range selects FS_SEL and GFS_SEL are 0 - 3, so 2-bit values are
left-shifted into positions 4:3
  uint8_t c = readByte(MPU9250_ADDRESS, GYRO_CONFIG); // get current
GYRO_CONFIG register value
 // c = c & ~0xE0; // Clear self-test bits [7:5]
  c = c & ~0x03; // Clear Fchoice bits [1:0]
  c = c & ~0x18; // Clear GFS bits [4:3]
  c = c | Gscale << 3; // Set full scale range for the gyro
 // c =| 0x00; // Set Fchoice for the gyro to 11 by writing its inverse to
bits 1:0 of GYRO_CONFIG
  writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c ); // Write new GYRO_CONFIG
value to register

 // Set accelerometer full-scale range configuration
  c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG); // get current ACCEL_CONFIG
register value
 // c = c & ~0xE0; // Clear self-test bits [7:5]
  c = c & ~0x18;  // Clear AFS bits [4:3]
  c = c | Ascale << 3; // Set full scale range for the accelerometer
  writeByte(MPU9250_ADDRESS, ACCEL_CONFIG, c); // Write new ACCEL_CONFIG
register value

 // Set accelerometer sample rate configuration
 // It is possible to get a 4 kHz sample rate from the accelerometer by
choosing 1 for
 // accel_fchoice_b bit [3]; in this case the bandwidth is 1.13 kHz
  c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG2); // get current
ACCEL_CONFIG2 register value
  c = c & ~0x0F; // Clear accel_fchoice_b (bit 3) and A_DLPFG (bits [2:0])
  c = c | 0x03;  // Set accelerometer rate to 1 kHz and bandwidth to 41 Hz
  writeByte(MPU9250_ADDRESS, ACCEL_CONFIG2, c); // Write new ACCEL_CONFIG2
register value

 // The accelerometer, gyro, and thermometer are set to 1 kHz sample rates,
 // but all these rates are further reduced by a factor of 5 to 200 Hz
because of the SMPLRT_DIV setting

  // Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
//   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit
0) interrupt
   delay(100);
}