kriswiner/MPU9250 #484: Azimuth

valon51, 2022-10-22:
Hi Kris, I would like advice on the correct formula for calculating Azimuth.. I have Euler Angels output from the dmp and when I tilt to the sides (- roll) the Azimuth fluctuates quite a bit, on the other hand it is not the case.. Otherwise, I have a hmc588l magnetometer.. Magnetometer I have them calibrated well.. Thank you in advance..

kriswiner, 2022-10-22:
I don't know anything about how the DMP works so not sure what could be the
trouble. Usually this kind of error is due to insufficient or improper
calibration. How can you be using the MPU9250 DMP AND the HMC5881
magnetometer? I do not understand this...

valon51, 2022-10-22:
no, I use mpu6050 together with hmc588l.. Pith and Roll are completely accurate and the Heading in the plane, and unfortunately I can't find the right formula anywhere that will cover me in all directions inclined heading. If the Roll goes into the minus, then the Heading fluctuates even by 20 degrees.. That's my problem, I'd rather help with the formula, because I'm 100% sure that the problem is only that.

kriswiner, 2022-10-22:
You need to make sure that the sensor data is properly configured, meaning
once you select which direction is N, then select your convention (either
NED, or ENU, for example) the sensor data are treated consistent with these
choices. So for NED, the accel/gyro data need to be treated as aN, aE, aD
and gN, gE, gD, and likewise this needs to align with the mN, mE, mD data.
This is opposed to trying to align the axes arbitrarily, which will cause
the kinds of behavior you report.

valon51, 2022-10-22:
As I said, I'm using data from dmp and the values ​​I measured really fit... I just think that I don't have the correct formula for calculating the virtual plane for the compass

valon51, 2022-10-22:
[image]

i still have a question i give this formula roll from 0 to 90 degree please..

kriswiner, 2022-10-22:
Is the order you are passing data to the Madgwick filter NED?

valon51, 2022-10-22:
I get data from dmp in the form of a matrix..

valon51, 2022-10-22:
Uploading received_464663602154494.mp4…

kriswiner, 2022-10-22:
In the Madgwick filter call you have to pass in the data as NED, so as aN,
aE, aD, gN, gE, gD, mN, mE, mD...do you see what I mean. I don't know how
to make it any clearer. If you do not, you will get very bad behavior, as
seen.

Do you aX, gX, and mX all point North?

valon51, 2022-10-23:
[image]
The problem was in the compass calibration, it was too slow and long, I used your calibration for the AK... It works with the HMC588L too, I just realized that it should take 15 seconds.
Well, since Matrix DMP is calculating me, I don't even have a way to see the values ​​or where to get them from.. I already have the output values ​​q1, q2, q3, q4.. At first I used Acc values ​​only without DMP, it had quite a lot of noise.. So used DMP here and the noise stopped, but I could see how I had problems with the compass itself. Here I give a photo of what my device looks like. Thanks..