kriswiner/MPU9250 #485: Computing Beta and Zeta Values

mecitpmk, 2022-12-12:
Hello , 
I was trying to tune the beta and zeta parameters to make the orientation better.

I was came across with those parameeters in the codes : 

```
float GyroMeasError = pi * (40.0f / 180.0f);   // gyroscope measurement error in rads/s (start at 40 deg/s)
float GyroMeasDrift = pi * (0.0f  / 180.0f);   // gyroscope measurement drift in rad/s/s (start at 0.0 deg/s/s)
```

How did you determine the gyroscope measurement error as **40 deg/s** ?
I couldnt see anything about it in the datasheet.
Besides that , same question for the  gyroscope measurement drift as **0.0 deg/s/s** .

If you have any knowledge about it please inform me about that.

Thank you,

kriswiner, 2022-12-12:
These parameters are irrelevant to orientation quality. Better to focus on
good calibration of the sensors, especially mag.

mecitpmk, 2022-12-12:
Thank you for the fast reply @kriswiner .

You are right, I should say the response time not the orientation quality.
Should I set those parameters with tuning and observing method to get better response time or its related with the datasheet informations?

Thank you,

kriswiner, 2022-12-12:
These are fusion filter parameters, and have nothing to do with the
underlying sensor properties. You could spend some time trying to adjust
these parameters but it is likely a waste of time. Better, after
achieving good calibration, to use a fast MCU. Something like a CM4F or
Teemsy 3.2 or 4.0 for example. Even an ESP32 would work well. This is
because the Madgwick filter is iterative and one should run the fusion
filter 10 - 20 times for every gyro data sample, and one should run the
gyro at 200 - 800 Hz. So slow MCUs will give very high latency....this is
much more important than these filter parameters....

mecitpmk, 2022-12-12:
Dear Winer,

Thank you for the repyling the content.

I was researching and going deeply into this Madgwick filter and I research iteration about Madgwick filter  and I found this :

According to Magdwick :
 "A conventional approach to optimisation would require multiple iterations of equation
(19) to be computed for each new orientation and corresponding senor measurements. Efficient algorithms would also require the step-size µ to be adjusted each iteration to an optimal value; usually obtained based on the second derivative of the objective function, the Hessian. **However**, these requirements considerably **increase the computational load** of the
algorithm and **are not necessary in this application**. _**It is acceptable to compute one iteration
per time sample provided that the convergence rate governed by µt**_ is equal or greater than the physical rate of change of orientation."

So I am confused about that. Do we **really** need to compute 10-20 times?

Thank you,

kriswiner, 2022-12-12:
yes, but try yourself to confirm

mecitpmk, 2022-12-12:
I will try and share the result.

Thank you,

mecitpmk, 2022-12-18:
I tried and come up with those result : 

Test 1:
  Assume that the sensor frequency = 1000Hz,
  Asseme that the Madgwick filter freq = 1000Hz,
  Result : Orientation filter response is much more clear and faster.

Test 2:
  Assume that the sensor frequency = 1000Hz,
  Asseme that the Madgwick filter freq = 2000Hz,
  Result : Orientation filter response is much more slower and delayed.

kriswiner, 2022-12-18:
Assume? Do you mean measure?

mecitpmk, 2022-12-18:
Yes, the experiment results based on the real IMU measurements.

According to my application, ( ** madgwick filter freq  = sensor frequency ** ) get me the best result.

Thank you,