kriswiner/MPU9250 #486: 9250 Who am I 0x75 ?

Wbiu, 2023-01-13:
Hello,

while the defaut value for the mpu9250 is 0x71, but mine ruturn 0x75 ?? 
My first though is of course that it is fake module, but Accel and Gyro seems to work fine or atleast the values are changing when I rotate the module. 

The who am i  Mag. in other hand returns 0xFF or 0. 
I´ve tried changing INT_PIN_CFG val from 0x12 and also to 0x02, but the Mag is still dead....

Any Ideal what i should try next? I´ve got 2 of these 0x75 module .... 

I´ve a working one to check if my sketches are working.
And of course my working one are retung 0x71 and 0x48.

kriswiner, 2023-01-13:
Likely you have a variant without the mag...since MPU9250 has been out of
production for a couple of years now it is not surprising people are
selling "MPU9250 Breakout Boards" with any available variants they can
find....

Wbiu, 2023-01-13:
Well thats unfortunate. I ordered 2x 9250s a few month ago and I recieved 1 with fully working (retuning 0x71 and 0x48.) and the other retunging 0x71 and FF or 0.

Since I need 2  9250s ,so I ordered  2 more for the higher chance to get a working one. 
And now i got 2  0x75....

Is there a safer alternative way to get a 9axis IMU?

kriswiner, 2023-01-13:
Order from reputable sellers? Might cost a bit more for quality though....

I have this
<https://www.tindie.com/products/onehorse/chainable-10-dof-motion-sensor/>
one available.

There are more accurate ones available too, like this
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/>
and
this <https://www.tindie.com/products/onehorse/usfsmax-module-and-carrier/>.

Maybe also try these
<https://sensormaestros.com/products/sensor/motion/9-axis/9-dof-9-axis-mpu-9250-breakout/>
guys?

Wbiu, 2023-01-13:
Alright thanks for suggestions.

Since I need a new one ether way. I might switch to BNO055 Adafruit, but let see and again thank you so far.

kriswiner, 2023-01-13:
I hope this works out for your application but in my testing the BNO055
does not work as well as the EM7180- and MAX32660-based boards I linked
to....

closed 2023-01-13 by Wbiu