kriswiner/MPU9250 #488: MPU9250 interrupt pin not working

XZhengLboro, 2023-04-05:
I have to configure the interrupts in the MPU set up to be active high, push-pull. However, I checked the oscilloscope that it is not working as expected.

Do you have any ideas about this issue?
Many thanks.

Below screenshot is the MPU9250 init.

[image]

XZhengLboro, 2023-04-05:
Hi Kriswiner,

I used the STM32H7 to connect the MPU9250.
I used the oscilloscope to check the MPU9250 INT pin, no high level output.
Also, I configured the GPIO of STM32H7 as GPIO_MODE_IT_RISING, and I checked the GPIO EXIT interrupt that is work. Please see the below figure.
[image]

Many thanks

kriswiner, 2023-04-05:
Usually, to get the MPU9250 to enable the interrupt, just those two
register writes to INT_CFG and INT_EN are required. This is what I do:

writeByte(MPUnum, INT_PIN_CFG, 0x10); // INT is 50 microsecond pulse and *any
read to clear*
writeByte(MPUnum, INT_ENABLE, 0x01); // Enable data ready (bit 0) interrupt

So if you do not select "any read to clear" you will have to manually clear
the interrupt, this is probably why it is not working. Check the data sheet
what 0x22 is selecting....you might be latching the interrupt or who knows
what...

closed 2023-04-06 by kriswiner