kriswiner/MPU9250 #489: Yaw values are not precise

taiebzbaiba, 2023-06-29:
Hi kris , I can't get precise yaw values on my mpu-9250 , I'm using Madgwick filter for sensor fusion , I tried changing sample rates parameters yet yaw values seem to drift for low sample rates values and for high sample rate I lose so much precision .

kriswiner, 2023-06-29:
The first question to ask is, how well are your sensors calibrated and how
do you know?

taiebzbaiba, 2023-06-29:
I tend to calibrate my sensors by measuring biases everytime I connect the device  and as for how I test it is I put my sensor on a certain position for a while , measure offsets so it becomes my zero , do random movements and rotations and then put the sensor back into that zero position .

kriswiner, 2023-06-29:
And how do you calibrate the mag?

kriswiner, 2023-06-29:
The next thing to check is what orientation convention are you using? NED
or ENU? Are you passing the scaled sensor data to the Madgwick algorithm in
the proper order for one of these?

taiebzbaiba, 2023-06-30:
> And how do you calibrate the mag?
> […](#)

I collect mag data for about 15 seconds , I rotate my sensor in every axis ,  I measure average mag biases and scale values in each axe based on collected data then store them for the main program

taiebzbaiba, 2023-06-30:
> The next thing to check is what orientation convention are you using? NED or ENU? Are you passing the scaled sensor data to the Madgwick algorithm in the proper order for one of these? On Thu, Jun 29, 2023 at 9:49 AM Tlera Corporation ***@***.***> wrote:
> […](#)

I guess you're referring to this issue https://github.com/kriswiner/MPU9250/issues/368 ,  I am passing these as parameters for my madgwick algorithm , I tried removing the (-) from mz out of curiosity and it gives way worse values.                                                madgwick(ax, ay, az, gx, gy, gz, mx, my, -mz, q);

kriswiner, 2023-06-30:
Yes, the issue is that the Madgwick algorithm expects the sensor values
that conform to the NED convention, so after you select which gyro axis you
want to be North, then pass the sensor data into the Madgwick function as
aN, aE, aD, gN, gE, gD, mN, mE, mD...the gyro and accel have the same axis
orientation but these are different from the mag. Once you get this sorted
out you should have much better results.

The last item is the speed of the sensor fusion relative to the sensor data
sample rate. For good results, the sample rate should be 200 - 500 Hz, the
low pass filter should be 0.1-0.2 x the sample rate, and the fusion rate
should be ~>10x the sample rate. So running on a CM4F or better, like a
Teensy 3.X or STM32L4/F4, etc

taiebzbaiba, 2023-07-11:
I tried ENU and NED , ENU seems to be more working for me , NED more drifty and I lose the origine (the zero value orientation) 
the problem for me is that for small angles it works decently , for big angles like 90 i get 130 , (always for yaw values)
Although I modified all sample rates , low pass filters according to your suggestion . As far the processor I'm working with ESP32-C3 32-bit RISC-V @160MHz.

kriswiner, 2023-07-11:
Not sure then what the problem might be. I would expect better than ~10
degree rms heading accuracy with proper calibration. Perhaps there are
environmental sources affecting the orientation estimation? Some sources of
emf like iron hardware or laptop, etc?