kriswiner/MPU9250 #491: Calibration - figure of eight

IanBUK, 2023-09-05:
I'm being a bit thick. My mpu9250 sensors look to be baldy calibrated and I'm wondering what I've done wrong. I read about the 'figure of eight' and am thinking 'wait....'. So silly question time: when moving the sensor in a figure of eight, do I keep it heading in the same direction keeping the yaw constant, or should I rotate it in the direction of travel, rotating through yaw? Should I be moving it flat on my desk, or vertically? Ok, that's two questions.

I might as well ask a third....when I flip the sensor through any axis, the reading take a *long* time to catch up. As in tens of seconds, not tenths of seconds. 

Ok, a fourth, and final question. If I leave the sensor alone, I'm seeing pitch drift over time. Why might this be?

I'd really appreciate some pointers on this, thanks.

Ian.

kriswiner, 2023-09-05:
So these problems are likely due to incomplete calibration and selecting
the incorrect reference frame.

For accel and gyro calibration, leaving the sensor flat and motionless on a
table top should be good enough fr reasonable accuracy. For the mag, the
idea is to sample the entire spherical response surface multiple times
during calibration. So imagine an arrow pointing up from the MPU9250
pckage. The goal of mag calibration is to point this arrow in every
possible direction (i.e., every point on the inner surface of a big balloon
surrounding the sensor)  multiple times. This usually takes a minute or two.

Then you need to pass the sensor data to the Madgwick or Mhony filter in
NED coordinates, meaning aN, aE, aD, gN, gE, gD, mN, mE, mD order (at least
for Madgwick). Could also use ENU I suppose. So the idea is to pick which
board direction you want to point to absolute North. Then identify which
sensor axes are aligned in this direction, there are then the N values.
Then once you know this, which are the E and D values? Pass these into the
fusion function and you should get resonable results.

The last part of the recipe is to use an MCU with enough power to sun the
fusion filter 10 - 20 times for each new gyro value. So if you run the gyro
at 200 Hz, run the fusion filter at 2 kHz or 4 kHz, etc.

IanBUK, 2023-09-05:
Thanks for the quick response. I shall have a go over the next few days, see where I get to and update here.Thanks,Ian.