kriswiner/MPU9250 #492: Reset FIFO and DMP

mark2b, 2023-11-12:
There is a line in the acc/gyr calibration function
`writeByte(MPU9250_ADDRESS, MPU9250_USER_CTRL, 0x0C);    // Reset FIFO and DMP`
Why value is 0x0C and not 0x6?
0x0C = 0b1100 
Bits 2 and 3 are high
But according to documentation for USER_CTRL, bit 3 is Reserved
[image]

kriswiner, 2023-11-12:
Long story short, Invensense didn't expose DMP control register bits in
their public MPU9250 documents. So bit 3 is DMP reset.

You can use 0x06, or try both and see what difference it makes (none on
power-up initialization), or consult the Motion_Sense code to see how
Invensense does this, or find a MPU9250 DMP application note to see how to
use this register bit, etc.

mark2b, 2023-11-12:
Thanks for quick answer and great repository

I have another question about calibration

I'm porting this code to rust for Pico 10DOF IMU
There is a code - writing accelerometer biases, for me it works when I comment out this code only 

`// Apparently this is not working for the acceleration biases in the MPU-9250
// Are we handling the temperature correction bit properly?
// Push accelerometer biases to hardware registers
  writeByte(MPU9250_ADDRESS, XA_OFFSET_H, data[0]);
  writeByte(MPU9250_ADDRESS, XA_OFFSET_L, data[1]);
  writeByte(MPU9250_ADDRESS, YA_OFFSET_H, data[2]);
  writeByte(MPU9250_ADDRESS, YA_OFFSET_L, data[3]);
  writeByte(MPU9250_ADDRESS, ZA_OFFSET_H, data[4]);
  writeByte(MPU9250_ADDRESS, ZA_OFFSET_L, data[5]);
`

kriswiner, 2023-11-12:
What is the question?

mark2b, 2023-11-12:
Sorry, should I expect it works properly if I write accelerometer biases or on some boards it will not work.
(on some **ino** files this code commented out)
I'm new with IMU.

kriswiner, 2023-11-12:
This technically works, that is, you should be able to write the accel
biases to the appropriate registers, but I have found the results to be
unsatisfactory since the LSB is serves double duty as some sort of
indicator bit such that the biases can be slightly off. So I apply accel
biases in the main loop. The gyro bias registers work fine.